ammolytics/projects #9: Setting up a "data is ready" interrupt

RipVW, 2019-02-21:


RipVW, 2019-02-21:
Oops - hit the enter key too soon.

Figuring this out took me a long time.  There are lots of examples for Arduino Unos and similar devices, but I didn't find much help for the Cortex MO.  Eventually I figured out the the Arduino Due, and (I think) Arduino Zero also have Cortex processors, so examples for those devices might also apply to the Feather.  Another, and embarrassing slow to materialize, flash of brilliance was when I started correctly spelling interrupt with TWO r's.  It's amazing how many more search results appeared!

Part of the frustration was that when it didn't work I didn't know which of the two devices I hadn't set up correctly.  Eventually it occurred to me that taking the jumper from the interrupt pin on the Cortex and tapping it on a ground pin would generate something like interrupt pulses and show if the Cortex was taking action when the pin went low.  One of those rare times when contact bounce is your friend.

Part of what I learned is that although just about any pin on the Feather can be used as an interrupt pin, multiple pins may trigger the same interrupt at the processor.  E.g. Pins 6 and A3 are both labeled EINT-4.
(https://cdn-learn.adafruit.com/assets/assets/000/046/195/original/Feather_M0_Adalogger_v2.2.pdf?1504805241)
So since I happened to use pin 6, I can't try to use pin A3 for another interrupt.  At least I think that's how it works.

In void setup() pin 6 is defined:
```
  // Pin 6 will be used to receive interupts from the LIS331HH accel when data is ready
  pinMode(6, INPUT);  // Pin connected to LIS331HH "INT 1" pin which is pin 3 on Sparkfun breakout board  
```
Your LIS3DH has a INT 1 pin, but I'm not sure where it shows up on the breakout board.

Then, after everything is set up and all is ready to take readings, here's the code for the Feather that tells it to pay attention to pin 6 and have the program jump to the interrupt service routine (ISR) whenever the pin meets certain conditions.
`  attachInterrupt(digitalPinToInterrupt(6), ISR_HH, HIGH);  // Enable interrupt`

As written it looks for pin 6 being HIGH.  I've found that LOW also works, and pissed away hours before giving up on RISING or FALLING working.  Have written Adafruit about that but so far there's no response.  Saw a lot of comments from folks having similar experiences.

ISR_HH is the name of the function I've defined elsewhere that reads the accelerometer:

```
void ISR_HH(){
  ISRaddr = HH;
  ISRdevice = 19;
  readAccel();
} // End of ISR_dataReady
```
readAccel() is a function I wrote to retrieve the two data bytes from the accelerometer.

It's considered good code hygiene to keep ISR's as short as possible, so ordinarily you wouldn't put a slow `Serial.print("Hello World - this is the ISR");` in a real ISR.  But it works to see if tapping a wire from your chosen interrupt pin to a ground pin triggers this ISR.

Maybe the library can be used to set up the accelerometer to send the interrupt when data is ready.   Here is how to do it manually on the LIS331HH:
```
  // Set CTRL_REG3 located at 0x22 to default values
  // This register sets up interupt parameters
     Wire.beginTransmission(HH);  // Talk to the 331
     Wire.write(0x22);  // Writes the address of the register
     // 1xxxxxxx 1 selects low on interrupt, default of 0 selects high on interrupt
     // xxxxxx10 Set bits to specify data ready signal on "INT 1" pin
     Wire.write(0x02);  // Writes into the register
     Wire.endTransmission(false); // false causes a restart to be sent  
```
Since I wrote 0x02 that means I left the most significant bit 0, which matches the HIGH parameter in the attachInterrupt statement above.  Setting the two lower bits to 10 says to put INT 1 high when data is ready.  I found this part of the datasheet to be less clear than usual.

On your 3DH the address of the register is still 0x22, but the register is organized quite a bit differently.  (See p.31)  It looks like setting the fourth highest bit to 1 will enable the data ready (I1_DRDY1) interrupt to appear on the INT 1 pin.  In other words write 0x10 to this register (00010000).

I hope you realize that you have to maintain this repository for as long as I'm alive.  I'll need to visit when  I've forgotten what the heck I did, or why I did it.

RipVW, 2019-02-21:
In my script I use a while statement to take a certain number of readings.  I put the attachInterrupt statement just before the while.  Then just after the closing } of the while I put ` detachInterrupt(6);`

RipVW, 2019-02-26:
Hooked up the 3DH tonight and can confirm that all of the above works and that writing 0x10 to CTRL_REG3 sets up the interrupt on INT1 when data is ready.

erichiggins, 2019-02-26:
Great!  btw, I learned why you were having issues creating branches. As it turns out, you may need to follow this workflow to create Pull Requests.
https://help.github.com/en/articles/creating-a-pull-request-from-a-fork

Hope it helps. Look forward to seeing what kind of sample rates you can achieve!

RipVW, 2019-02-26:
Thanks.  Also want to thank you for the DEBUG idea.  Was getting ready to comment out lots of Serial.print statements to enable battery operation.  Read up on how your DEBUG strategy worked and added that idea to my script - a big time saver.  Also switched from SD.h to SdFat.h so that I can use the YYYYMMDDHHMM filename format like you do.  That way if I ever find my rtc chip, crystal, etc. the script will be ready.

One time I was looking at your script and got the impression you communicate with the accelerometer using SPI.  Is that right, or do you use I2C?

erichiggins, 2019-02-26:
I'm using I2C though I may switch it over to SPI today just to see if there's any performance increase.

RipVW, 2019-02-27:
Would be interested to know how to set up another SPI.  I don't think I2C is your primary bottleneck at this time, but every place you can find a uSecond helps.