ammolytics/projects #14: Add iOS support for Open Trickler app

erichiggins, 2019-05-01:
The Open Trickler is available for Android, but not yet for Apple devices.

Apple requires jumping through a few hoops, such as:

- Signing up for their [Developer Program](https://developer.apple.com/) ($99/yr)
- Using the [Apple ID associated with their Developer Program as the primary account](https://developer.apple.com/support/account/authentication/) on a development device (iPhone, iPad)

erichiggins, 2019-06-21:
Blocked by https://github.com/pauldemarco/flutter_blue/issues/276

erichiggins, 2020-01-04:
A build of the iOS app is working and available through Test Flight.
Anyone interested in trying it out can get access here:
https://testflight.apple.com/join/1XSoUPL3

Shooter0423, 2020-01-23:
Like the IOS keypad better, weight unit didn’t seem to display correctly on my screen, just saw a “g” assuming it should 2 characters. The value changed so it was obvisously doing the math for the conversion from grams to grains.

erichiggins, 2020-01-23:
The app supports both grams (g) and grains (gr), the button toggles between them. In the redesign, this will be more obvious.

erichiggins, 2020-05-07:
I've added the Open Trickler app back to the Apple TestFlight store:
https://testflight.apple.com/join/1XSoUPL3

Shooter0423, 2020-05-07:
Thanks @erichiggins, the link says its good for 90 days.
To the Group
Am I the only one using an Apple device. I use it on an old phone, an iPhone 5.

erichiggins, 2020-05-07:
@Shooter0423 No, you're definitely not the only one using it.