ammolytics/projects #18: Mobile app does not update target weight when BT connection is re-established

spacegunr, 2019-05-09:
When the BT link is dropped (after Pi reset for instance), and then re-established, the target weight has not changed on the mobile app.  It appears that the Pi believes the target weight is zero (default value), and it will not start working until the target weight is changed on the app, at which point I think the target weight is re-sent to the Pi and then it starts to trickle.

erichiggins, 2019-05-09:
Thanks for filing this! I've noticed this as well -- I've worked around it by just hitting the + button then the - button to re-register the target weight with the controller.

One scenario I'd like to get your thoughts on:
If the mobile app is left running and then opened the next day for a different loading session, should it still show the last-set target weight? At what point should the target weight be reset on both the app and controller?

spacegunr, 2019-05-09:
Your welcome :)  It was as much for you as it was for me to learn the process of how to file a bug ticket.

I would say let the mobile app keep the weight setting, but push that value back to the Pi once a BT connection is (re)-established.

erichiggins, 2020-07-27:
Hello! I created a Discord to make it easier for folks who are building an Open Trickler to chat with and support each other.

https://discord.gg/Bx3ynT

This invite link expires in 1 day to keep out spammers, but feel free to email me directly and I'd be happy to invite you!