ammolytics/projects #21: Add powder-drop support

erichiggins, 2019-06-05:
Some folks have suggested adding support for an automatic powder drop, controlled by a servo. Anyone who has experience with this can chime in here with suggestions.

Since it's an optional feature, it should be treated as such in the software (controller and Mobile).

Passaita, 2020-04-10:
While I can't help with the code I think code can be universal and hardware (servo) optional with the following algorithm. When scale is stabilized at zero or above and regular trickler cycle starts system could send on GPIO 17 or 27 PWM servo signal from 0 to 90 to 0 (someone help here. I don't have powder dispenser and not sure what is the lever move in degrees). Then continue to execute trickler code.
Things to consider:
- Servo specific code (some servos are mirrored i.e. position 180 on one can be 0 on the other)
- Can mess with small loads (then just disconnect powder-drop)
- Some servos prefer incremental PWM so cycled from 0 to 90 in steps of 5 with the pause in between will help
- Add enable / disable powder drop in the app
- Add reverse servo movement in the app
- Add min / max servo values in the app

erichiggins, 2020-05-30:
I believe @MrMr-No did something similar by [calling some Python code](https://github.com/ammolytics/projects/issues/28#issuecomment-583269764) they wrote. Perhaps they can chime in here to share their code so that it can be added to the project?

MrMr-No, 2020-06-05:
[steper.py.txt](https://github.com/ammolytics/projects/files/4734605/steper.py.txt)
 case (tickDelta > 1499 && this.scale.weight === 0  ) :
        
        console.log('*******Steper Python code******')
        console.log(`Current speed ${tickDelta} `)
       let {PythonShell} = require('python-shell')
PythonShell.run('/home/pi/projects/trickler/peripheral/lib/steper.py', null, function(err) {
  if (err) throw err
  console.log('finished stepper')
})

MrMr-No, 2020-06-05:
I modified tricckler.js file to execute steper.py script (attached above) when you need more then 30 grains of powder  in same directory as trickler.js

MrMr-No, 2020-06-06:
[image]
Here u have how to connect stepmotor drv 8825 to pi

erichiggins, 2020-06-06:
@MrMr-No This is excellent, thank you!

MrMr-No, 2020-06-06:
in acction
[https://rapidgator.net/file/f219cb8233dc6f66fba69269c4b7b101/IMG_0077.MOV.html
](url)

erichiggins, 2020-06-08:
Embedding the Python code shared above to avoid the extra download:

```
from time import sleep
import RPi.GPIO as GPIO

DIR = 20   # Direction GPIO Pin
STEP = 21  # Step GPIO Pin
CW = 1     # Clockwise Rotation
CCW = 0    # Counterclockwise Rotation
SPR = 220   # Steps per Revolution (360 / 7.5) 96

GPIO.setmode(GPIO.BCM)
GPIO.setup(DIR, GPIO.OUT)
GPIO.setup(STEP, GPIO.OUT)
GPIO.output(DIR, CW)

step_count = SPR
delay = .0208 / 4

for x in range(step_count):
    GPIO.output(STEP, GPIO.HIGH)
    sleep(delay)
    GPIO.output(STEP, GPIO.LOW)
    sleep(delay)

sleep(.5)
GPIO.output(DIR, CCW)
for x in range(step_count):
    GPIO.output(STEP, GPIO.HIGH)
    sleep(delay)
    GPIO.output(STEP, GPIO.LOW)
    sleep(delay)

GPIO.cleanup()
```

TQBCPC, 2020-11-27:
> I modified tricckler.js file to execute steper.py script (attached above) when you need more then 30 grains of powder in same directory as trickler.js

Can you please share your modified trickler.js file?

boki1397, 2021-02-13:
@MrMr-No Hello, 
your video is impressive! and your drawing is very good. 
which stepmotor did you use?
it is possible to make the complete modified image available here? 
unfortunately i have no idea about the programming or software operation in raspberry.
it would certainly be of great help to many like me.
many thanks for your effort.

MrMr-No, 2021-02-13:
Tnx! Unfortunately, Isn’t so simple. I can post image, but I have designed completely new Hornady’s powders rotor to working with bipolar step motor like You can see on video. Perhaps  is better solution to waiting on @ erichiggins new project to rewrite code to Python to see what we can do.

boki1397, 2021-02-14:
@MrMr-No 
Thank you for your prompt reply. 
i watched the video several times and i saw the 3d parts. 
i have a 3d printer and i think i can copy it.
if you think it would be better to wait for the python programming, then i will be patient.
thank you for making your knowledge and work available to the general public.
I also like to help others. unfortunately that's not my area of ​​expertise.