ammolytics/projects #28: Installation Problems

GTMux, 2019-10-25:
This project is great! And I would love to bring it in our shooting club for these people who like accurate reloads. Unfortunately I'm not able to get the RPi Zero running due to problems during the installation process:

.
.
.
.

npm install

> bluetooth-hci-socket@0.5.1 install /home/pi/node_modules/bluetooth-hci-socket
> node-gyp rebuild

make: Entering directory '/home/pi/node_modules/bluetooth-hci-socket/build'
  CXX(target) Release/obj.target/binding/src/BluetoothHciSocket.o
In file included from ../src/BluetoothHciSocket.cpp:8:
../../nan/nan.h: In function ‘void Nan::AsyncQueueWorker(Nan::AsyncWorker*)’:
../../nan/nan.h:2298:62: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
     , reinterpret_cast<uv_after_work_cb>(AsyncExecuteComplete)
                                                 
...
...
...
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:198:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Linux 4.19.75+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/node_modules/bluetooth-hci-socket
gyp ERR! node -v v10.17.0
gyp ERR! node-gyp -v v5.0.5
gyp ERR! not ok

erichiggins, 2019-10-26:
@GTMux Thanks for reporting this!  If you have the complete installation log available, would you mind including it here in the ticket, or as a separate Gist? I think some critical information from the trace may be missing.

Lastly, can you provide a little bit of detail about your setup?

- Host machine (Mac, Windows, Linux):
- Raspbian version:

Thanks again!

GTMux, 2019-10-26:
Hi Eric, 
thanks for the fast response. Setup is: 

- a Raspberry Zero W (armv6l)
- installed from an iMac (High Sierra / 10.13.6) using balenaEtcher to transfer the Raspbian image  (Raspbian Buster Lite, Version:September 2019, Release date:2019-09-26, Kernel version:4.19)

System setup started based on your description (rename, expand etc., update, install bluetooth bluez libbluetooth-dev libudev-dev, cloning project)

And than this time it seems to work fine except some warnings concerning missing pre-build resources (please see InstallLog.txt attached). Only the service installation ended with a failure (Log attached). And unfortunately I got an error message in the app on my android tab that no device is connected.

The difference between today and the last 5 days of installation trials: I only did:
sudo apt-get update 
and NOT sudo apt-get upgrade 
and NOT sudo apt-get distr-upgrade

Maybe you have some hints for me to get this running?

Thanks for your time!

[2019-10-26_InstallLog.txt](https://github.com/ammolytics/projects/files/3774511/2019-10-26_InstallLog.txt)
[2019-10-24_ServiceInstfail.txt](https://github.com/ammolytics/projects/files/3774512/2019-10-24_ServiceInstfail.txt)

GTMux, 2019-11-08:
Anyone have any ideas?
I would be very happy about advice. :) 

Thanks and best regards,
Ralf

erichiggins, 2019-11-08:
Sorry for the delay (I was traveling for work)!

I have some time this weekend to dig into this issue -- I'd first like to see if I can reproduce it.

The real solution, I believe, would be for me to provide a ready-to-flash image that has everything installed. That avoids this type of complication.

GTMux, 2019-11-08:
Thanks a lot - this would be great. I was close to ask you for an image. :)
Sorry to bother you with this problem...

erichiggins, 2019-11-15:
Just a quick update on this: I have made some significant progress on a pre-configured image. It's nearly there, just requires a few more tweaks. Once it's ready, I'll share a link for you to give it a try. If it works for you, I'll update this repo and docs w/ an official link to it.

erichiggins, 2019-12-07:
Another update: I know it's taking a while -- sorry about that! It's **very** close now.

More context: I'm using a system called [buildroot](https://buildroot.org/) to build a custom SD card image. [The manual is long and complex](https://buildroot.org/downloads/manual/manual.html), and it takes a few hours to run a complete build from scratch. So, it's not quite as easy as making a small change then being able to test it quickly. I still have to write the image to an SD card, boot a system, and see if everything is working.

It's coming along, but is very, very time-consuming. The two pieces I'm blocked on are:

1. Getting bluetooth working correctly
2. Getting the Open Trickler software to run correctly on boot

Thanks for your patience on this. If you happen to have any experience with buildroot, help is always appreciated!

GTMux, 2019-12-07:
Hi Eric,

thanks for the update. I keep my fingers crossed that you succeed. 
Unfortunately, I have no idea how I could help with software related stuff...

erichiggins, 2019-12-14:
Great news! I just got bluetooth working!
It's late now, so I haven't tried loading the OpenTrickler software yet. I'll give that a shot tomorrow -- it's looking very promising so far!

erichiggins, 2019-12-15:
SUCCESS! I'll share an image file today along with some instructions. It'll help if someone can try it out and make sure it works on their end before I publish it to the world.

GTMux, 2019-12-15:
great - would be a pleasure for me to test it! :)

erichiggins, 2019-12-16:
[Test image file can be downloaded here](https://drive.google.com/open?id=1uTVCknwQI6sZAX_cFkh84gF3V91xCv71). Please be sure to read all of the following instructions and notes.

Please let me know if you have any questions, have any issues, or (hopefully) are able to get it working!

## Instructions

- Download the attached SD card image file  
  shasum: `47910ee0389418c11f7afd112ec4d9cad7d5ce06`
- Flash the image to your microSD card (I highly recommend [balena etcher](https://www.balena.io/etcher/))
- If you want it to be able to login via SSH over WiFi (optional):  
  Edit `boot/wpa_supplicant.conf` and add the SSID and password for your wireless network.
- Insert the microSD card into your Pi, connect it to your scale, and boot it up
- After about 60 seconds, try to connect to it with the [OpenTrickler Android App](https://play.google.com/store/apps/details?id=com.ammolytics.opentrickler&hl=en_US).

## Notes

- It's designed to use the PWM pin **`12`** on the Pi, rather than pin `15` which I used originally. You can change this in the `ecosystem.config.js` file on the `code` partition if needed (more below).
- It takes about 60-90 seconds for the system to fully boot and the OpenTrickler to be visible to the Android App.
- The microSD card will have three partitions: `boot`, `code`, and `opentrickler`. The first two are formatted with FAT32 and can be viewed/modified from Windows and Mac computers.
- The source code from this repo and all dependencies are contained on the `code` partition, so you can modify easily and try your own version.
- There are no logs available in this test image. I'm working on that.
- The attached file is compressed with [`xz`](https://en.wikipedia.org/wiki/XZ_Utils) to reduce the `.img` file from 320MB to 52MB. You *might* need to decompress it before flashing.
- It will not show up as `opentrickler.local` on your network. You'll need to check your WiFi router to figure out it's IP address in order to SSH into it.
- You can SSH into it using `root` as the login. The password is `ammolytics`

erichiggins, 2019-12-18:
Quick update: I'm working on another image that addresses some of the missing features that I listed above. Namely:

- Addressable on the network at `opentrickler.local`
- Improved startup performance
- Logging (if connected to the device via SSH)

GTMux, 2019-12-18:
Hi Eric,

Thanks for your updates. I downloaded the image, decompressed it, brought it on the SD card using balena etcher with success. Due to the fact that my scale is currently borrowed by a friend I was not able to connect the scale. 

Currently I can't establish a bluetooth connection and can't find the raspberry via bluetooth. Open trickler app: You are not connected to a device.

Is it necessary to connect to a scale first?

matsyngvelindgren, 2019-12-18:
IT WORKS SUPER NICE !!!
Awesome job man!!!
Many thanks and a big hug :-D

erichiggins, 2019-12-18:
@GTMux From the Android App, you need to connect to the unit. You do **not** need to pair with the app from your Bluetooth menu/system settings.

- Open the menu (top-left corner of the screen)
- Select "Bluetooth Devices"
- Tap the button on the bottom-right of the screen and it should try to connect
- On successful connection, you should see the readings of the scale on screen
- Then, go back to the home view and you can control the trickler

I'm pretty sure I demonstrated all of this in [my YouTube video](https://www.youtube.com/watch?v=1BODqGHfVtk).

Hope this helps!

FWIW: The UI needs some updates to make this all easier based on what we've learned from using it.

GTMux, 2019-12-18:
@erichiggins Thanks for the information but this is what I did. The app isn't showing any device. Of course I tapped the button on the bottem-right of the screen a couple of times. :) Maybe because of my old Android Tab. I will try with a newer model in the following days.

[image]

erichiggins, 2019-12-18:
@GTMux That's interesting. A few things come to mind:

The Pi won't advertise itself on Bluetooth if it cannot detect the scale. The scale has to be plugged into it, turned on, and correctly configured.

Assuming a correct setup on the Pi/Scale, the question is: Is the problem the Pi or the Android app?
There are a few ways to find the answer to this. There's a free Android app called [nRF Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en_US) which is pretty useful for debugging Bluetooth devices. It will let you see every device that is advertising.

cyclepath37, 2019-12-21:
I was having the same issue as GTMux and then noticed only the boot partition was on the sd card. I’m not sure what i am doing wrong.

erichiggins, 2019-12-21:
@cyclepath37 Which operating system are you using? Windows and Apple computers won't be able to see all three, but the `boot` and `code` should both be visible.

As long as your using the balena etcher program to flash the SD card, it should create the same result.

Did you configure your scale using the instructions listed in my article?

cyclepath37, 2019-12-21:
Windows and using balena etcher as suggested. Only showing boot if i etch with or without decompressing. 
I have an A&D HR202i, but have it configured the same as you show with the fx120.

erichiggins, 2019-12-21:
@cyclepath37 I'm not familiar with that particular model, so you may want to double check the manual to verify that it's RS232 interface operates the same way. For the sake of this conversation, let's assume it does for now.

Potential causes:
- Incorrect serial to USB adapters (they aren't all the same, even with identical physical connections)
- Scale not plugged in and turned on before booting the Pi
- Driver on the Pi being flakey. I've seen this on occasion and there's not a clear software workaround. Just have to reboot the Pi and try again

I can share another image tomorrow that includes logging, which would help to diagnose these problems more easily.

With the image you have, you can configure it to connect to your WiFi then SSH into the Pi and see what's going on, if you're comfortable working with the Linux command line.

Shooter0423, 2019-12-27:
@erichiggins thanks for the updated image. I loaded to two different RPi Zero W's with different brand SD cards (1 16g, the other 32g) both connected to the scale  and worked to ready the scale value in the startup are. My issue comes in 2 parts.
1) After the initial success with the 2 setups, they developed a inability to connect, it was getting late so I shut everything done. tried it just now and things are working fine again, don't know what I did different? Is there a correct way to disconnect
2) Havent been able to get the vibro motor to do its thing. I know the motor and tube work cause I can test the on an Arduino. Is there a reason that the control pin changed, don't think it is the problem but just asking

erichiggins, 2019-12-27:
@Shooter0423 Glad you were able to get them working fairly easily -- now to see if we can figure out some of the issues you've run into.

Let's start with the motors not running. The reason I changed the control pin from 15 to 12 was to allow the use of PWM (Pulse Width Modulation) control on the motors.

Here's a [pin out diagram for the Pi Zero W](https://pinout.xyz/pinout/io_pi_zero). Note that physical pin 15 is BCM 22 and physical pin 12 is BCM 18 (PWM 0)

You can check if this is the issue you're having pretty easily with two different methods:

First, power up the OpenTrickler Pi & scale in the usual way, connect w/ the app, then set a charge weight (like 10.00 gr) and hit the Play button. At this point, the motor should be running. If it's doing nothing, you can check the control pin on the Pi using a multimeter/voltmeter. You should see it turning on/off with approx 3V. If you don't, then check the other pin.

The other method you can use is to change the pin setting in the config file. Pull the microSD card out of the Pi and connect it to your computer. You should see a drive called `code`. On that drive, there will be a folder named `opentrickler` which has a file named `ecosystem.config.js`. Open that file with a text editor (NOT a word processor). You can change the `MOTOR_PIN` setting from `12` to whatever you are using. You may still want to use a multimeter in the same way I described above to verify that it's working as expected.

As for the issues you're having with disconnecting... I may need more info for that. There could be a software bug that I can fix. For now, how about we get the motor working first?

Shooter0423, 2019-12-28:
Thanks for the quick response. I had debugged the issue the way you described but the real culprit turned out to be a bad solder joint. So everything is up and running. The Bluetooth connection seem to need more time than I’m allowing, I think that will eliminates the problem, I’ll let you know.

erichiggins, 2019-12-28:
@Shooter0423 Great! If you are able to reproduce the bluetooth issues, go ahead and open a new issue here on GitHub so we can track it separately.

@GTMux @cyclepath37 Have either of you had luck getting yours working?

GTMux, 2019-12-29:
@erichiggins: thanks for the update. As mentioned I‘m still waiting for the scale. When the scale arrives back at my home I‘ll let you know directly.

cyclepath37, 2019-12-29:
Still stuck. I confirmed the fx & hr202i balance RS232 pins are the same. Does it need the entire breadboard motor circuit connected for the bluetooth to work? I am using an old dandy trickler and haven’t tied it in yet, so i have been trying to connect with just the bare pi zero without the motor circuit.

erichiggins, 2019-12-29:
@cyclepath37 The Pi doesn't try to detect the motor circuitry at all (not even sure if that's possible), so you should be fine without it. The Pi only waits for a scale to be detected and the bluetooth drivers before it advertises itself so that the app can connect to it.

Did you [follow these instructions to configure your scale](https://blog.ammolytics.com/2019-04-30/diy-smart-trickler.html#prerequisite-scale-configuration)? Without those steps, the Pi won't be able to communicate with the scale. It's not just as easy as plugging it in, unfortunately.

Datakalu, 2019-12-31:
Having the same result as GTMux and cyclepath37. Flashed the test image using Windows & balena, fx120 connected and configuration confirmed. Pi does not advertise on Bluetooth.

Shooter0423, 2019-12-31:
Seems like I'm having a little better go at it then @GTMux,  @cyclepath37 and @Datakalu. I'm working with 2 different RPi Zero W's, one has been problematic from the beginning in regards to the Bluetooth connection. #1 would connect on startup only some of the times, like 1 in 10 on average. #2 will connect on startup most of the time, but sometimes it may take 2 or 3 startups (once as many a 5 this evening). I haven't isolated anything in my work area that could causing interference, (for a short time both where working kinda consistently and then my wife would come in the room and they would quite, I sure was going to miss her).
My startup procedure and some observations.
I quite plugging the power in and out on the pi and use the switch on a power strip, not sure how robust the micro usb sockets are.
I wait 2 minutes after turning on power to the pi to boot, I set a timer.
I then start the mobile app, wait a little, go into the menu and get Bluetooth Devices, wait a little, then hit the green button.  If it doesn't come back with the info from the scale in 10 or 15 seconds, it not going to because there is no connection ( I always wait for the app to tell me so ) try the green button a time or 2 more (never seems to work a second time if it didn't work the first) then turn off power, check the scale, check the cables, check that i'm not going crazy and do another startup.
I know its frustrating, but this thing works.
It looks like @erichiggins is using a stick powder in the video and I am working with load of Tac under a 77gr SMK, (also got a Varget load for a 6BR to work on) I haven't loaded real ammo yet, probably do that tomorrow as it looks like Wednesday will be the best day to go out this week.  I built my vibro-motor different so I'm tuning the motor.js and probably will need to reimplement some speed steps that have been commented out in trickler.js.  I'll start new threads with this info as I get it organized.

chaseerickson12, 2019-12-31:
@erichiggins I’m having the same problem as everyone above. @Shooter0423 does the pi require a 16 or 32g sd? I’ve tried a 4 and a 8g and can’t get anything to work. I’m going to pick up a 16g after work and a usb to micro adventure adapter and see if that makes a difference over using a micro sd to sd adapter. Is the any way to test to make sure your on the right trail.

Shooter0423, 2019-12-31:
@chaseerickson12 I was trying to run 2 pi's. One has a 16g standard Kingston, the other a 32g Ultra Scandisk. I swap the cards back and forth in the good working pi, as they both have the new image as supplied by @erichiggins recently. I use the adapter that comes with a micro card, have several laying around, don't think that matters. Anyway in my case it doesnt matter. Are you downloading the compressed image and using Belena to flash it to the card? Earlier when I was having problem with the first pi, a site suggested formatting cards with a freebie called 'SD Card Formatter', don't know that it actually fixed any problems but I use it every time before flashing cards now. Another suggested freebie, same site I think, is Note++. Works will when modifying/editing files in the code folder

erichiggins, 2019-12-31:
Thanks to everyone for keeping me posted -- I know it can be frustrating but it really does help a lot to get feedback like this so that I can focus on the right problems that will work for everyone. Also thanks to @Shooter0423 for chiming in, helping others, and sharing your experience!

I think the biggest issue is that **we're flying blind**.

The only debugging info you have available is whether or not the app can connect to the Pi over bluetooth. There are a lot of things that could go wrong to prevent that, but without having some way to see that, it's impossible to find the root cause.

I suspect that if I can make the Pi's logs available via your web browser at http://opentrickler.local on your local Wifi, then we'd at least have a better idea of what's going right and what's going wrong, like:

- If the Pi is booting up at all.
- If the Bluetooth drivers are failing (I've seen this)
- If the Open Trickler program (on the Pi) is crashing and not restarting
- If the Pi isn't seeing the scale.

The other issue that I think may be worth addressing is allowing the Mobile app to connect to the Pi **even if the scale isn't connected**. This requires work on the mobile app as well, so it's a bit more work to pull off -- because of this, I've been kind of avoiding it.

I'd like to get everyone's thoughts on this though before I proceed. You might have better ideas and I'd love to hear them!

erichiggins, 2019-12-31:
@chaseerickson12 A 4GB or 8GB card should be fine. It would probably even work on a 1GB card, though I haven't tried it.

Shooter0423, 2019-12-31:
Thanks @erichiggins, glad to help. I think viewing a log will pay dividends faster. The mobile app is working good enough.

Datakalu, 2019-12-31:
I have the wrong USB to Serial cable. I suspect that’s my problem and I’ll let you know how it plays out when I have the correct cable.

chaseerickson12, 2020-01-01:
@Shooter0423 I have not been decompressing the file. Ill give it a shot. Im interested to hear if @Datakalu serial cable was the problem

chaseerickson12, 2020-01-01:
https://www.amazon.com/gp/product/B075XHWVSJ/ref=as_li_qf_asin_il_tl?ie=UTF8&tag=ammolytics0f-20&creative=9325&linkCode=as2&creativeASIN=B075XHWVSJ&linkId=7c662ec9d4021bf3c1374f86ff1b9b0d 

Do we need to be using a usb to serial that requires this adapter? The one I purchased from the recommended list didn’t require this to be plugged in. Or could this be the problem that I’m experiencing

erichiggins, 2020-01-01:
@chaseerickson12 Yes, you do need that actually. I know that you can connect the cables without it, but the adapter doesn't change the male/female interface, it changes the pin wiring arrangement. The A&D scales use a rather strange wiring setup which requires this in order to work.

chaseerickson12, 2020-01-01:
@erichiggins i think I’m slowly eliminating my problems one at a time lol 😂. I’ll report back now that I’m trying the adapter

chaseerickson12, 2020-01-01:
@erichiggins i can get Bluetooth connection 👍🏻 Now time to figure out where my wiring is messed up lol so the trickler works

erichiggins, 2020-01-01:
@chaseerickson12 let's follow up on that over email tomorrow. Send me photos of what you have and I'll see if I can help sort it out.

erichiggins, 2020-01-04:
# Update
I've made the logs available via web browser on the OpenTrickler with the following uSD image for the Pi. After you flash the uSD card with the following image, you will need to modify the `boot / wpa_supplicant.conf` file with the details for your WiFi network so that the Pi can connect to it.

Once booted, you can view the logs by visiting [`http://opentrickler.local/`](http://opentrickler.local/) in the web browser of any computer or mobile device that's on the same WiFi network. If it does not resolve, the most likely issue is a misconfigured `wpa_supplicant.conf` file.

**Do not share the contents of your `wpa_supplicant.conf` files here, as they will contain the password for your WiFi network and it may compromise your network security.**

Documentation for `wpa_supplicant.conf` can be found online. [Here's one example](https://linux.die.net/man/5/wpa_supplicant.conf).

## uSD Pi Image
This image is considered alpha quality and is intended only for testing purposes until these issues can be resolved and verified by all interested parties. Afterward, I'll provide some cleanups and a new image, which everyone should expect to download and reflash their uSD cards with.

[Download `opentrickler-1.1.6-ALPHA.img.zx`](https://drive.google.com/open?id=1H1-Y0sm0sSx40kNqHaOhjIgBHQxGmJH6)

## Checksums
Optional, in case you want to verify the image after downloading and extracting
- `sha  bf78327d5f9768637e6877cc8da8ccf3efe56fd7  opentrickler-1.1.6-ALPHA.img`
- `sha256  3bc9e5bb5a02d10dd160d68631198c8a7edd073a952ca8abe481bcecbaa8c797  opentrickler-1.1.6-ALPHA.img`
- `md5  149cb844703f2bec4e954ef5296bf449  opentrickler-1.1.6-ALPHA.img`

Datakalu, 2020-01-04:
Haven’t flashed the latest image but I’ve install the correct serial cable and was able to connect on a couple of occasions. However, it drops the connection before I can run the motor circuit.

Datakalu, 2020-01-04:
Motor circuit runs, turns out it was me when returning to the main screen. Haven’t thrown powder yet but works as described. Bluetooth connection is hit and miss.

erichiggins, 2020-01-04:
@Datakalu The logging capability in the latest image should give more insight as to why you may be having Bluetooth connection issues. I've seen a couple occasions where the USB/Serial driver acts strangely.

Datakalu, 2020-01-04:
@erichiggins I’ve never played around with tech projects like this and it’s been an education getting to this point. I’ll keep slowly chipping away at it but I’m learning as I go. Great project and support, thank you!

MrMr-No, 2020-01-08:
Can we get  Pi password?

erichiggins, 2020-01-08:
@MrMr-No Yes! I actually shared it in [an earlier post on this thread](https://github.com/ammolytics/projects/issues/28#issuecomment-566235134).

cyclepath37, 2020-01-13:
Update: I continued to struggle with bluetooth connection loss as many others had also. It would connect, then drop and sometimes take several attempts to reconnect, followed by dropping it again. I ended up ordering a different serial cable. 
It must have been the cable because the new cable has connected right away with no loss after running it for 20 minutes!!!!
Here is the cable I ordered 
USB to RS232 Adapter with PL2303 Chipset, CableCreation 3.3 ft USB 2.0 to RS232 Female DB9 Serial Converter Cable for Cashier Register, Modem, Scanner, Digital Cameras, CNC, 1M Black https://www.amazon.com/dp/B0769FY7R7/ref=cm_sw_r_cp_api_i_EHphEbVHZDVP9
Up and running!!!!

Shooter0423, 2020-02-03:
Update:
I have been having intermittent connection problems. Most of the time the Bluetooth would connect but the trickler program and the scale would not report TRUE and the program would say in a loop and never connect. I would cycle the power to the RPi and let the program start over.  Generally after 2 or 3 tries it would connect and all would be good. I followed @cyclepath37 cable suggestion from above, (I did not replace the USB to micro USB adapter) and been starting and restarting over the last couple hours, probably 10 or 12 times by now. The RPi has connected everytime.

erichiggins, 2020-02-03:
@Shooter0423 Interesting! Thanks for the update! I'll order one of these cables and try it out, then update the parts list. So happy that the community was able to find a great solution for this one!

Shooter0423, 2020-02-04:
@erichiggins 
I think the cable definitely fixed my connection issues. More starts and restarts today, working every time. I have the led code to monitor startup embedded in the Index.js file and theled's will even blink on in different orders as the program is retrying to establish connections. Interesting situation, if you forget to turn on the scale, the BT will connect, program and scale will report FALSE, turn on the scale and both will report TRUE and OpenTrickler is ready to roll.
I need to post something about the edits to the code and layout on the proto-bonnet. A picture will not be very good, and a hand sketch probably worse, have any suggestions. And what to do with the modified files.
[image]
Pic show a drv8825 stepper driver, i got that mostly working, not sure best place to activate the powder measurer, looking at the code in trickler.js where it starts the motor when the scale is zero for more than 1 second

Datakalu, 2020-02-06:
I ordered the Cable Creations cable as well and I am longer experiencing Bluetooth issues. So far it’s been 100% and connecting quickly. Thank you @cyclepath37 & @Shooter0423 !!! Moving forward and starting to build a Dandy style trickler for the system.

Shooter0423, 2020-02-06:
@Datakalu great news, haven't heard any news from @erichiggins about his experience with the new cable. We owe a bit of thanks to @cyclepath37. If you start a new thread about your trickler build, i'll post a couple pictures of a trickler I built out of gray pvc conduit connectors (iirc about $5 maybe $7)

MrMr-No, 2020-02-06:
I  activate stepmotor only when I need need more then 30 grains 

 case (tickDelta > 1499):
        this.motor.speed = SPEEDS.VERY_FAST
        
...
....

drv8825 working very well :)

erichiggins, 2020-02-06:
I ordered a new cable, but I'm out of town for work and won't get a chance to try it until tomorrow.

Thanks to everyone for sharing their feedback & suggestions, generally helping each other out and being awesome!

erichiggins, 2020-02-06:
@Shooter0423 Two ways to share your code changes:

1. You can create a public Gist here on GitHub. Very easy.
  https://gist.github.com/
2. You can create a [Pull Request](https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/about-pull-requests) to add your changes into the repo. I can review it, suggest changes, then merge it.

Don't hesitate to email me if you need some help.

Shooter0423, 2020-02-06:
@MrMr-No, great idea to put it in as another speed option.  I have struggled to keep it from double throwing in shouldGo or startWhenReady.  Are you setting up a function for the stepper code and calling it and are you using setIntervals or msleep/usleep.  My coding skill with js is poor, any code you want to share will be greatly appreciated.

[17 more replies]

closed 2020-05-30 by erichiggins