ammolytics/projects #30: Start Up Log/Status Indicator

Shooter0423, 2020-01-05:
Have a questions about the new image, and a question about an Improvement?
1) New image allows for changing password, restoring network security as before?
2) With the log file I can see 3 status situations that have to be true to work correctly. A) TRICKLER, B) SCALE and C) BT. I found 3 variables that are set to TRUE, TRICKLER_READY, BT_READY, TRICKLER.scale.ready when ready. I assume the event handles can detect when they are changed to TRUE. If 3 additional pins (7, 13, 18) where defined in ecosystem.config.js and their OUTPUT set to HIGH when the variables are set to TRUE then we can know without looking at the log that things are good.  I know real estate is at a premium on the perma-proto bonnet but 3 leds (and accompanying resistors) could be added to give a visual indicator. And maybe we need a variable SCALE_READY to keep the coding looking consistent?

erichiggins, 2020-01-05:
Thanks for the suggestions!

> New image allows for changing password, restoring network security as before?

You can actually already change the root password now, if you wish, by running the `passwd` command on the Pi.

Example terminal output (as run from an OSX or Linux computer):
```
$ ssh root@opentrickler.local

root@opentrickler.local's password: ....
# passwd

Changing password for root
New password: 
Retype password: 
passwd: password for root changed by root
```

But your point is still a good one. The production image will likely just disable logging into the device entirely to help lock it down, but I'll also offer a development image for the tech savvy folks who want to tinker.

> I know real estate is at a premium on the perma-proto bonnet but 3 leds (and accompanying resistors) could be added to give a visual indicator.

The status LEDs are a good idea! If that's something you want toprototype on yours, I can assist you with the code portion. If you can get it working well, then we can incorporate it into the project!

The more community support for features like this, the easier it is for me to maintain and focus on other things. The goal of Ammolytics isn't just to design and support DIY projects for free 😂

Shooter0423, 2020-01-05:
I will work on the status leds, I see how you did the pin assignment in the config file, probably put the script in discrete  file and use ‘require’ . I am slowly figuring this out, good simple place to start. Got some ideas for a powder drop also.

Shooter0423, 2020-01-23:
I have had success with implementing status LED's. 3 led's and current limiting resistor fit pretty easily on the perma-proto bonnet. I edited the ecosystem.config.js file to declare the pin assignments and modified the index.js file in the "lib" subfolder to initialize the pin LOW and then set to HIGH when TRICKLER_READY is TRUE, TRICKLER.scale.ready is TRUE and BT_READY. Currently this only indicates what happened at startup and is not checked afterwards.

erichiggins, 2020-01-23:
@Shooter0423 Great! If you'd like to create a pull request and some wiring advice, I can review and add it to the project for others to benefit from!

ande7824, 2020-03-10:
Hi Eric,
I have everything working with your alpha image, but the trickler is severely overshooting (0.2-0.5gr) for each fill.  I can't seem to edit the motor.js file because its listed as ready only.  However I am logged in as root and the permissions seem to show that I have read and write permissions.  Can you help me out? I can't figure out what I am missing. 

I am editing with vim and through putty.

Thank you

erichiggins, 2020-03-10:
@ande7824 The OpenTrickler code that lives on the `CODE` partition is read-only from the perspective of the Pi. This is intentional for a few reasons, but mainly so that nothing about the Pi can alter change the code that needs to be run.

If you remove the SD card and put it into your computer, you will be able to modify it. That is the intended approach, since you can make changes on your computer then copy them over to the SD card to test. This way, if you happen to wipe the SD card by accident, you'll still have a backup.

Regarding overshooting, I have two suggestions:
1. **Prime the trickling tube on your OpenTrickler**  
  The goal here is to get powder into the tube so that trickling happens at a consistent rate. I like to set my usual target weight through the app then place an empty pan onto the scale. This will cause the OpenTrickler to run at it's fastest speed first, priming the tube, and eventually slowing down until it reaches the target weight.
2. **Adjust the trickler angle**  
  The angle of your powder trickler plays a significant role in tuning both the speed and precision of the OpenTrickler. Too steep towards the pan and it will trickle too fast, causing over-pours. Too shallow and it will run very, very slowly. Especially for the last few kernels of powder.  
  Same as before: Set the usual target weight and keep filling the pan, adjusting things as necessary, until you get a few perfect throws in a row.

I have to make minor angle adjustments on mine when switching between H4350 and Varget, which are both stick powders that look nearly identical. They sure don't trickle identically though.

ande7824, 2020-03-10:
Hi Eric- great suggestions that I hadn't even thought of from the physical trickler setup perspective.

I did try to modify the SD card files but Windows keeps saying I need to format the card :/. I may just try reflashing it with some speed changes and try to standardize the angle and initial priming process to see if I can get things to improve.  Thanks for the great blog post and quick reply!

TQBCPC, 2020-09-13:
> I have had success with implementing status LED's. 3 led's and current limiting resistor fit pretty easily on the perma-proto bonnet. I edited the ecosystem.config.js file to declare the pin [...]

@Shooter0423 Shooter0423 what resistirá and LED did you use?

@erichiggins, can you update the SD Card image with this feature?

erichiggins, 2021-02-05:
Quick note about this ticket. It's pretty easy to control the built-in LED on the trickler. My question is, what would make sense for using it for status? A pre-defined series of blinks? Solid vs blinking? Pulsing? Fast vs slow blinking? A combo of all of these?

erichiggins, 2021-08-15:
I've updated the trickler to change the built-in LEDs with status. This behavior can be changed/modified, and other LEDs can be used simply by modifying the included [`opentrickler_config.ini`](https://github.com/ammolytics/projects/blob/develop/trickler/peripheral/opentrickler_config.ini) file.

closed 2021-08-15 by erichiggins