ammolytics/projects #35: Use other RS232 scales i.e. Acculab

ian99rt, 2020-04-09:
I'd like to branch off and make the Pi code work with an Acculab V-1mg over a similar but slightly different RS232 protocol. 

What files and variables would control both the RS-232 comms setup and message definition / decoding?

erichiggins, 2020-04-09:
Great! Actually you don't have to branch at all, the intent is for the Open Trickler to support a variety of scale brands & models, but it requires contributions from folks like you!

Use the code in [`and-fxfz.js`](https://github.com/ammolytics/projects/blob/develop/trickler/peripheral/lib/and-fxfz.js) for reference and create your own version, providing the same variables and methods used by other files (such as `index.js` and `trickler.js`. I can work with you to add support for your Acculab rather than having to creating a fork for different scales!

One of the tricky parts is detecting which type of scale is being used and loading the right code file for it, but it should be solvable.

Hope this helps!

ian99rt, 2020-04-10:
Awesome Eric! thanks for adding the label too. I'm a newbie to github. I'm completely out of the loop when it comes to javascript. I'll do my best to read thru the code and try and figure things out. Any way I can communicate with you that won't be all over github to view my ignorance?

erichiggins, 2020-04-10:
This is a great place for the communication because it's easier to maintain context, share code snippets, and get help from others working on similar things. That helps me out because it can be challenging to keep up with everyone, I end up being a bottleneck whereas on here, anyone can chime in and help out. Nobody will shame you here, especially for trying to learn something new!

That being said, my email address is listed at the bottom of all every article on my website, so you're welcome to send one over.

ian99rt, 2020-04-10:
http://photos.labwrench.com/equipmentManuals/355-5466.pdf

This is where i'm seeing the info for the acculab serial comms.

erichiggins, 2020-04-10:
Hm. I haven't worked with this scale myself and the documentation doesn't make it very clear how it actually works, so you may be in for a steeper learning curve.

The library I'm using to work with serial port communications is this:
https://serialport.io/
The documentation is pretty good, so you shouldn't have too much trouble.

If you're more familiar with another programming language than JavaScript, you might try working with those first just to see if you can read from the scale over the serial port. Then it can be a bit easier to translate to another language or library.

ian99rt, 2020-04-10:
Eric I was going to work with raw comms over something Serial and the PC to create a road map of commmands. It doesn't look like it have alot of controls over RS-232. 

Their description of the controls seems odd to me as well but this seems to be the list of available controls. This scale does stay on the last unit of measure used so I don't think we'd need to swap back and forth between units. 

Definitely doesn't look like the scale can be polled for its model and S/N like it looks like your code does. I'd assume if your code sent unknown messages to the scale it would simply ignore them. 

The question is: will your Pi code error out if it doesn't get all the information it polls the A&D scale for?

[image]

[image]

ian99rt, 2020-04-10:
I've also go a friend who is a computer/software engineer to help out a bit. Surprisingly hes not on github but maybe soon.

ian99rt, 2020-04-10:
Well I may have goofed. The scale I bought on ebay doesn't go into grains (only g, oz, ct, dwt) ..... Kind of annoying but not end of the world. When you switch between grains and grams on the android app you showed on your video that it automatically converts between the two hopefully that will still function without doing lots of manual conversion.

erichiggins, 2020-04-10:
> Eric I was going to work with raw comms over something Serial and the PC to create a road map of commmands. It looks like there is not alot of controls over RS-232.

That's a great approach! From what I can see the commands it provides should be fine. All you really need is the current weight and if the scale is stable. If it sends the unit of measure as well, that's a bonus, but you can work around it.

> Their description of the controls seems odd to me as well but this seems to be the list of available controls. This scale does stay on the last unit of measure used so I don't think we'd need to swap back and forth between units.

That is interesting. In your case you may want to just leave it on grams, which will offer the best resolution anyway.

> Definitely doesn't look like the scale can be polled for its model and S/N like it looks like your code does. I'd assume if your code sent unknown messages to the scale it would simply ignore them.

That's ok for now. It may make it a little more complicated to determine which brand of scale is connected but it's more of a problem if 100 were supported than just 2.

> The question is: will your Pi code error out if it doesn't get all the information it polls the A&D scale for?
It doesn't really _need_ most of that information, so you can just remove it in yours.

erichiggins, 2020-04-10:
> When you switch between grains and grams on the android app you showed on your video that it automatically converts between the two hopefully that will still function without doing lots of manual conversion.

The scale does most of the work on that front, actually. The conversion is simple arithmetic, so not a huge problem though there are gotchas when writing code that deals with decimals.

erichiggins, 2020-07-27:
Hello! I created a Discord to make it easier for folks who are building an Open Trickler to chat with and support each other.

https://discord.gg/Bx3ynT

This invite link expires in 1 day to keep out spammers, but feel free to email me directly and I'd be happy to invite you!