ammolytics/projects #39: Connect to mettler toledo

RowanNL, 2020-06-23:
Hello guys,

I really like this project! I’m a electrical engineer but Unfortunately Im very bad in programming, I’m more a 400VAC guy... Can someone help me to re-write the code for a mettler Toledo balance? It uses MT SICS

[MT SICS - datasheet](https://www.mt.com/mt_ext_files/Editorial/Generic/7/MT-SICS_for_Excellence_Balances_BA_Editorial-Generic_1116311007471_files/Excellence-SICS-BA-e-11780711B.pdf) 

Thanks!

With kind regards,

erichiggins, 2020-06-24:
Thanks for linking to the docs, that makes a huge difference!

I've seen a lot of requests for a lot of different scales, which makes me think that I should add some simple scale debugging mode that people can turn on and just see what's coming over the wire so that code can more easily be written to adapt to different brands and models.

RowanNL, 2020-06-25:
It's the least i could do...

I've found also something else on github, maybe it helps?     [https://github.com/janelia-pypi/mettler_toledo_device_python](url)

kind regards,

erichiggins, 2020-07-27:
Hello! I created a Discord to make it easier for folks who are building an Open Trickler to chat with and support each other.

https://discord.gg/Bx3ynT

This invite link expires in 1 day to keep out spammers, but feel free to email me directly and I'd be happy to invite you!