ammolytics/projects #41: Can't view logs in browser

ande7824, 2020-08-10:
Issue: When I view the hyperlinks in the opening page, both links come up with just blank pages.

Build information:
-  I can't access the open trickler with "http://opentrickler.local" but I can access via the IP shown on my router, so I am not sure if this is part of the issue or not.
-  I am using the most recent DEV image flashed with Balena Etcher.
-  I am using the same components linked in your build

Any ideas what my issue is? I would like to see the logs because I am trying to dial in my motor speeds.

Thank you.

erichiggins, 2020-08-10:
If you can't access the opentrickler.local website, then there are a few things it could be.

- WiFi not configured on the device (you said you can access by IP, so this isn't it)
- Not connected to the same network (if you phone is using cellular but not WiFi. Also not the case for you)
- Mobile device using VPN (this may prevent the domain name from resolving as it changes your DNS settings)
- Mobile device using alternate DNS (Google: 8.8.8.8, CloudFlare: 1.1.1.1, etc). You'll need to use your router as the DNS server in order for it to see devices by domain name on the local network

It's good that you can get to the debug page that has the links for the log viewers! If the log pages are blank, first give them a few minutes. Occasionally, things can be slow to boot up. You don't need to refresh them, they stream the log contents automatically as the logs on the device change.

Hope some of this helps!

ande7824, 2020-08-13:
Hi Eric,
OK, some additional info.
- I do have more than one network, but I am accessing via wifi on the same network as the open trickler with a Dell laptop.  
- I changed the network to a different one on the opentrickler and cannot access the logs
- I tried with chrome and microsoft edge both with "http://opentrickler.local" and the assigned IP address.  For both browsers, I cannot access with "http://opentrickler.local" but I can with the IP address
- I accessed via IP and clicked the hyperlinks and waited 15 min in both browsers, but still saw blank pages.

Any other ideas?  I mean this isn't a deal breaker by any means, but it would be nice to see the logs to help me understand what is happening. I am not a coder by any means.  

Thanks for all the help and this project in general!  I really like my trickler so far.

erichiggins, 2020-08-13:
Interesting. We may need to walk through this. [Join us on Discord](https://discord.gg/WqTbyK2), where there are several other folks building Open Trickler units and helping each other out.

erichiggins, 2020-08-14:
@ande7824 If you were able to resolve this, would you mind adding notes about what you needed to do to fix it?

ande7824, 2020-08-15:
I fixed it by adding a fixed IP for the open trickler in my router. Then I added opentrickler.local and the assigned IP to the client host file.  I followed instructions here:  https://www.siteground.com/kb/how_to_use_the_hosts_file/
I did the fixed IP as a result of a suggestion while googleing this issue but can't find the page again for why.  I am fairly savvy with computers but networking is out of my comfort zone so not even sure if the fixed IP is needed.

closed 2020-08-15 by ande7824