ammolytics/projects #42: Write OpenTrickler FAQ

erichiggins, 2020-08-10:
Many of the questions for the OpenTrickler could be answered in an FAQ.

Common items:

- Not showing up on BT (bad adapter, scale not connected/detected)
- Support for other scales
- Accessing the CODE partition
- Will do you make it for Arduino/ESP/etc
- Powder drop support
- Accessing the log web page  #41

ande7824, 2020-08-13:
This may help someone, but I figured out how to access the CODE partition in Windows 10 by utilizing the Windows disk utility with the SD card in my computer.  The CODE partition shows up in the card but didn't have a drive letter assigned so I right clicked and assigned a letter and it showed up right away.

erichiggins, 2020-08-14:
Added here:
https://github.com/ammolytics/projects/blob/develop/trickler/README.md#frequently-asked-questions

closed 2020-08-14 by erichiggins