ammolytics/projects #43: Mock interface does not work

alien2108, 2020-08-19:
https://github.com/ammolytics/projects/blob/c9a3dc6b3b893de39ef160e6f5b9179b719764e2/trickler/peripheral/lib/mock-scale.js#L165

/home/pi/trickler/peripheral/lib/mock-scale.js:165
    this._unit = trickler.TricklerUnits.GRAINS
                                        ^
TypeError: Cannot read property 'GRAINS' of undefined
    at new MockScale (/home/pi/trickler/peripheral/lib/mock-scale.js:165:41)
    at new SerialPort (/home/pi/trickler/peripheral/node_modules/@serialport/stream/lib/index.js:145:19)

Indeed nowhere in code TricklerUnits is even defined:

pi@raspberrypi:~/trickler/peripheral/lib $ grep TricklerUnits *
grep: build: Is a directory
mock-scale.js:    this._unit = trickler.TricklerUnits.GRAINS
mock-scale.js:    this._unit = trickler.TricklerUnits.GRAINS === this._unit ? trickler.TricklerUnits.GRAMS : trickler.TricklerUnits.GRAINS
mock-scale.js:      case trickler.TricklerUnits.GRAINS:
mock-scale.js:      case trickler.TricklerUnits.GRAMS:

erichiggins, 2020-08-19:
This is my fault. I added this mock interface so the developer who helped create the mobile app had something to work with (because they don't own a scale). I have not maintained the mock interface, so I'm not entirely surprised that it broke.

I've been rethinking the workflow though, so I'd like to hear your thoughts.

Right now, Bluetooth doesn't advertise unless the scale is working. Seemed like a good idea at the time, but now I think it causes more confusion than it prevents. I think I might change that -- advertise on BT and just surface the scale status to app so users know where the issue is.

Additionally, there are other brands/models of scales that folks would like to use instead. To make it easier to develop a new scale adapter, I might write a generic/boilerplate one that simply logs whatever it reads from the serial port and has stubs that can be filled in to make it functional.

Neither of these solve the mock scale being broken, of course. I'm wondering if it's still needed or useful if I made the two changes suggested above.

alien2108, 2020-08-25:
I would rather suggest you do the following:

- Make bluetooth optional and implement direct functions to be called (with bluetooth disabled):
a) setWeight
b) getWeight
c) getStatus
d) mainLoop (here you place code that should be executed all the time as refresh display, read keys...)

Basically all you display on bluetooth phone I would like to display on a simple Pi Zero Display HAT as https://www.waveshare.com/wiki/1.44inch_LCD_HAT

It has joystick (up/down +-1gr, left/right +-0.1gr), three buttons (1 - get default weight, 2 - set default weight, 3 ...) and a cool display. No bluetooth, no phone...

I tried to implement it, but your code currently is too interlinked with bluetooth.

erichiggins, 2020-08-25:
I see. Does this relate to the mock scale issue somehow? If you're not planning on using Bluetooth, then this falls a bit further outside of the Open Trickler intended design.

Since it's an open source project, you're welcome to fork and modify the code to meet your use case. A lot of folks have different design ideas that they share with me, but I cannot commit to writing several different versions of this project to support them.

We have a lot of folks discussing their designs and ideas over on [Discord](https://discord.gg/WqTbyK2). Perhaps you can discuss it there and get some support from the community!

erichiggins, 2023-03-27:
Closing this out since the code has changed significantly.

closed 2023-03-27 by erichiggins