ammolytics/projects #47: Trickler doesn´t work

boki1397, 2020-09-25:
Hello at all.
I´m a greenhorn with raspberry and programming and i need help.

first I set up and built everything according to the instructions.

i have connected the raspberry to the app.

after entering the weight and the start button, the trickler does not run.

if I drive the diode directly with electricity, the motor runs

i have no idea where the error is.

can someone help me?

I can also send the raspberry logs.

sorry for my bad English. I'm from Germany

Thank you for your help in advance.

erichiggins, 2020-09-25:
Hey @boki1397 !

It's possible that there's an issue which pin you're connected to for activating the motor.

If you'd like to join us on Discord, there are a lot of other people who have built Open Tricklers that can help out by sharing their experience. I think we may even have a few other German-speakers in there too!
https://discord.gg/WqTbyK2

Otherwise, perhaps you can share a photo of your build so I can see which pins you're connected to on the Raspberry Pi?

erichiggins, 2021-08-15:
Has your issue been resolved?

erichiggins, 2023-03-27:
Marking this as closed since I haven't heard back.

closed 2023-03-27 by erichiggins