ammolytics/projects #59: Raspberry Pi 3 B and G&G JJ200B scale

patrickk83, 2022-04-10:
Hi ladies and gentlemen!

I want to share the experience to get the code running on a Raspberry Pi 3 B in combination with a G&G JJ200B laboratory scale (German manufacturer). 
Eric was a great help with the code on the Pi. More informations follow if someone is interrested.

Regards

Patrick

patrickk83, 2022-04-10:
Manual of the JJ200B (only available in German):
[JJB.pdf](https://github.com/ammolytics/projects/files/8459683/JJB.pdf)

erichiggins, 2022-04-12:
Another version of the manual for English speakers
https://www.gandg.de/download/anleitungen/englisch/JJ-B%20Englisch.pdf

HayaminiNL, 2022-08-09:
Hi Patrickk83, i am interested in setting up the Pi3B with opentrickler.
I am struggling with it for a couple of nights now and did not succeed yet.

I am using the A&D FX120i and already pulled this repos to the Pi3B, because any other Pi-board is hard to get at the moment and the Pi3B I have laying around and it was collecting some dust until a couple days ago.

Thank you in advance!

HayaminiNL, 2022-08-09:
> Hi ladies and gentlemen!
> 
> I want to share the experience to get the code running on a Raspberry Pi 3 B in combination with a G&G JJ200B laboratory scale (German manufacturer). Eric was a great help with the code on the Pi. More informations follow if someone is interrested.
> 
> Regards
> 
> Patrick

Hi Patrickk83, i am interested in setting up the Pi3B with opentrickler.
I am struggling with it for a couple of nights now and did not succeed yet.

I am using the A&D FX120i and already pulled this repos to the Pi3B, because any other Pi-board is hard to get at the moment and the Pi3B I have laying around and it was collecting some dust until a couple days ago.

Thank you in advance!

erichiggins, 2022-08-12:
@HayaminiNL Join us on Discord to discuss this and get some support!
https://discord.gg/WqTbyK2

erichiggins, 2023-03-08:
In progress: https://github.com/ammolytics/open-trickler-peripheral/pull/7